tscircuit/tscircuit.com #3: Multiple lock files in codebase

DhairyaMajmudar, 2024-10-01:
-> There are two lockfiles in the project codebase which are `package-lock.json` generated due to npm and `bun.lockb` generated due to bun

-> Having these two different lock files in the codebase causes PacakgeManager related issues so deleting one of them would be good to go
 
-> ScreenShot from VSCode 

[image]

DhairyaMajmudar, 2024-10-01:
POV: Removing package-lock.json would be a good decision to be taken reason being bun is faster in installing dependencies as compared to npm 

What's your POV on this @seveibar ?

imrishabh18, 2024-10-01:
@DhairyaMajmudar Good point! Yes, we can remove the `package-lock` file. I see @Complexlity has raised a PR for it as well.

Thanks for pointing this out

closed 2024-10-01 by imrishabh18