tscircuit/tscircuit.com #7: Make "New" dropdown work

seveibar, 2024-10-01:
[image]

[image]

DhairyaMajmudar, 2024-10-01:
@seveibar I would like to take this up can you pls. assign

seveibar, 2024-10-02:
@DhairyaMajmudar @Complexlity just nabbed this. Issues can go fast!!!! There are many other issues available if you just click around the UI it's very broken

closed 2024-10-02 by seveibar

DhairyaMajmudar, 2024-10-02:
> [@DhairyaMajmudar](https://github.com/DhairyaMajmudar) [@Complexlity](https://github.com/Complexlity) just nabbed this. Issues can go fast!!!! There are many other issues available if you just click around the UI it's very broken

@Complexlity super fast man! NW I'll look of other things : )