tscircuit/tscircuit.com #16: Download buttons should work

seveibar, 2024-10-05:
- [ ] Download TSX should work
- [ ] Download Circuit JSON should work

imrishabh18, 2024-10-06:
@seveibar By download you mean that a `.tsx` file should be downloaded and similarly a `.json` for circuit?

seveibar, 2024-10-06:
@imrishabh18 thinking about this more, I don't think we should support that as a feature because snippets will eventually be able to have multiple files and people can literally just ctrl-c/cmd-c in the short term. I removed that checklist item

imrishabh18, 2024-10-06:
Yeah, I thought by download you mean't to download the file. Which is not required as you can copy paste

Abse2001, 2024-10-09:
@seveibar  Im going to fix this issue to be clear no .tsx file because people can just copy and paste right ?

seveibar, 2024-10-09:
Yep!

closed 2024-12-14 by Abse2001