tscircuit/tscircuit.com #24: Edit snippet name

imrishabh18, 2024-10-06:
Currently, all the snippets are having the same `untitled` name and no option to edit it

PALLAVIKHEDLE, 2024-10-07:
@imrishabh18 Can you please assign it to me?  I will finish it quickly.

imrishabh18, 2024-10-07:
Sure! Thanks for picking this up

PALLAVIKHEDLE, 2024-10-07:
@imrishabh18  - Do you mind running through the steps on how to create snippets? I am not able to create one/reproduce the use-case. Thanks!

imrishabh18, 2024-10-07:
As you can see in the video, that there is no option to edit the name `untitled-board-3` snippet. Probably one should be able to edit the name of the snippet they have. Somewhat like a pencil icon on the side of the name, to enable edit.

cc: @seveibar 

https://github.com/user-attachments/assets/c27ce3bc-6fb9-433b-a659-ebb0a28e34a8

PALLAVIKHEDLE, 2024-10-07:
Thank you so much! I will make changes and submit it for review in morning!

PALLAVIKHEDLE, 2024-10-07:
[image]
I am not able to generate a code. API Failed.

seveibar, 2024-10-07:
@PALLAVIKHEDLE this is fixed on the `/editor` page if you pull main. To use AI, you have to follow the instructions in the README to get an ANTHROPIC_API_KEY, reach out to me on discord if you need one

PALLAVIKHEDLE, 2024-10-07:
Thanks for getting back to me! 
I need an API key, 'pallavi3014' this is my discord ID. I am sorry not able to join you on discord.

PALLAVIKHEDLE, 2024-10-08:
@seveibar, can you please provide an API key?

seveibar, 2024-10-09:
@PALLAVIKHEDLE you can join the discord at this link, DM me there: https://tscircuit.com/join

seveibar, 2024-10-14:
This is now fixed via the editor page #57

closed 2024-10-14 by seveibar