tscircuit/tscircuit.com #25: Mobile view Is broken

valentimarco, 2024-10-06:
When opening the site on mobile, the view Is broken!
A list of issues:
1. The size of the Page Is actually this but only of you zoom out (i am not in desktop mode, as brave calls)
[image]

2. View when opening the First time:
[image]

3. Text overflow in buttons:
[image]
[image]

4. List of trending out of screen and Page size much bigger than screen size
[image]

seveibar, 2024-10-06:
Great issue, totally agree we need to fix this!

DhairyaMajmudar, 2024-10-06:
@seveibar would love to take on this can you pls. assign

seveibar, 2024-10-06:
because this issue is ELITE (chance of failure is high) we won't assign, but feel free to go for it, we just don't want someone to grab it and not finish

seveibar, 2024-10-06:
@DhairyaMajmudar fwiw you could break this into sub-issues that are assignable

DhairyaMajmudar, 2024-10-06:
> [@DhairyaMajmudar](https://github.com/DhairyaMajmudar) fwiw you could break this into sub-issues that are assignable

Yup, much better

PALLAVIKHEDLE, 2024-10-07:
@seveibar Can you please assign it to me? I will finish it quickly.

closed 2024-12-26 by github-actions[bot]