tscircuit/tscircuit.com #90: Add a Loader for Better User Experience During Content Loading

KhushiParekh, 2024-10-18:
Description:

Currently, there is no visual indicator when content is being loaded except the loading text that too has a slight delay, which may cause users to feel that the application is unresponsive during these times. I suggest adding an animated loader (or a simple loader) to enhance the user experience by providing a smoother and more intuitive feel.

By implementing a loader, the users will have a clear indication that something is happening behind the scenes, which will improve the perceived performance of the app and reduce the potential for confusion.

Suggested Changes:

Introduce an animated loader or a simple loading spinner during data fetches or other loading states.
Ensure the loader is displayed consistently across different parts of the app where loading occurs.
The loader should disappear as soon as the content is fully loaded.

I would love to take on this task and contribute the application. Could you please assign this issue to me?

imrishabh18, 2024-10-18:
Sounds good to me. Please go ahead and make a PR

seveibar, 2024-10-18:
This already happens on some pages, i think maybe the editor page has it? Could use that as a reference.

Minimal implementations are preferable because when we switch to SSR for pages there is no loader.

closed 2024-12-26 by github-actions[bot]