tscircuit/tscircuit.com #94: Add favicon from tscircuit.com

seveibar, 2024-10-18:
So that an icon appears in the tab

AlexVCS, 2024-10-18:
hi there, I'd like to work on this issue. can you please assign it to me?

AlexVCS, 2024-10-18:
I added this simple code to `index.html` but I can't get `bun run dev` to run without an error after installing the dependencies.

Here's the line I want to add (and check to see that it works):
`<link rel="icon" href="/favicon.ico" type="image/x-icon" sizes="16x16">`

This is the error I get when I run `bun run dev` (this isn't the full error but should be enough for you to get the gist):
```
✘ [ERROR] Could not resolve "./dist/bundle.js"

    vite.config.ts:8:29:
      8 │ import winterspecBundle from "./dist/bundle.js"
        ╵                              ~~~~~~~~~~~~~~~~~~

failed to load config from /Users/alexcurtisslep/Documents/projects/snippets/vite.config.ts
error when starting dev server:
Error: Build failed with 1 error:
vite.config.ts:8:29: ERROR: Could not resolve "./dist/bundle.js"
    at failureErrorWithLog (/Users/alexcurtisslep/Documents/projects/snippets/node_modules/esbuild/lib/main.js:1472:15)
```

Any suggestions?

seveibar, 2024-10-21:
@AlexVCS sorry for the delay, i don't know why i didn't get the notification but they will come through if you mention me directly i.e. @seveibar . Sorry about that

You just need to run `bun run build` to resolve that error you're seeing.

Best of luck!

seveibar, 2024-10-21:
Actually are you on windows? It kind of looks like the "fake api" that's used by snippets to emulate the api isn't building.

You can explicitly build that with `bun run build:fake-api`. It should create the `dist` directory, if it doesn't then that's the underlying issue

AlexVCS, 2024-10-21:
> bun run build:fake-api

I'm on Mac but running that command fixed the issue @seveibar ! PR incoming

closed 2024-11-02 by seveibar