tscircuit/tscircuit.com #128: Error:  port pin U1 pin1 has multiple pcb matches unclear how to place pcb port smtpad 1 smtpad GND smtpad

nischalj10, 2024-10-28:
[Snippet code to reproduce](https://snippets.tscircuit.com/editor#data:application/gzip;base64,H4sIAP1/H2cAA+VcW2/jNhZ+968g/LCYARxFFKlbMO7uorvYDbBNBhh00aIoAkVWJmpjy5DkdqZB/ns/3jROHE40YIqGqh4C8ugc6vt4SJqXE9brbdP25JaUbVX01bdd9XUD0aba9OSOXLXNmsz/0Xdl3Za7uj8um7aaz7RR/3FbwRIG62bzv+Jjs+vfts22e8xwK17MZ7Oy2XQ92dbQv6xuOrIktzNC5nR+Qub/OfvXfCFy0b0cEzn2f6ZyXOROz0OVi1WOqlyicpHKpSqn7TKV4yqXq1yscjRU2URnJZrT81RnJZzT80xnJZ7T81xnNSCUofIGksZENSiUovIaFopReQ0M5ai8hoZyVF1obChH5TU4lKPyGh3KUXkND+WovMaHclRe40M5Km8qzZSv8XFTvsbHNN5I42MaL9P4uP4+k/jgYCSUQAIUAl0DKEkLdJEoSgtMmRKjEOhKYBKkEGiUTKIUAg2TSZhCYHBInEKgK4JLoBAgoQQGKRJKYJAioQQGKRJKYJAioQQGKRJKYJAioQQGqalQVJQWaKTcIEVCNWqDFAklMEiRUAKDFAklMEiRUAKDFAklMEiRUAKDFAklMEiRUAKDFAklMEiRUB3OIEVCCfY68h0pOiL7+6def1asK9Hpzy9/qso++KW42VXdq2E4eD2b1Zu+aq+KsiJqLKk+9NVm1T0yyoiBY4PyTkjXt/Xm/exuNqs+yJFJfe/f796y6OIdu/jm9Oz0gl6cZfjyKzkSnajSX5PlV7Kctup37Ya8QpKQN+V1vZUpQm6DIJAWd1pQFqtvmlV1s7wVduppLn/6tr05GfKoiOu+x1eOj9dCt1xtgmEoDMpmfVwV3cdqVVzI193xqvl1c9MUq7/vdvVqSaM0vIzLq7QoIp5elfkqLKo0L2ieZEnOw79tN8uvo5yyKFSuU0/b9EVfN5vzq6uu6k8wMn84IeGCfJR/f8NfcvdJe9t09VjtO0N+cNTydkiad115/bbevNsWJXyxvA2DNN5/Bbf8s22Lzftqjd+W/dq7qa76d/UKfvwkI2RVt2ghAIj21Dfbo745umz6vlnvMZaA4Msf9iSE0AWJFoQtCF+QeEGSBUkXJFuQfEEoqFG8p1Cg0KBQodChUKLQolCj+f4HCIlgE4kyYRPBJoJNBJsINhFsIthEKJuF+3Y/fsrsVXpbv79+gqsiKeiC9ZNcE4CL8fEYIGKAiQEqBrhYcAfYGKBjgI+hx6HHocehx6HHocf5fbJcVBtsOGw4bJggBhsGGwYbBhuGshn0mKhl+hTpoe10u+32pq7at0Xbn+3Wl1WLVrTXCsquBLv50LCHwoYSrpqm36Kno/UMZm8G4Vd7QN50635brO5REwPDf6GHr/6A+caPplT9trz8bjk/SoI8jUP0MfVQtl7PH+p9v5zHQR6HYZrk6knih2q/1qv+ejkPAyYVqFbMc/5Q87oSbUKopuJ9pgzE81Czuy621XIumsr+i+MvIh65Eqd4mUc2jC+WN3PjzYMIDo+5hpj643DuRpwF0KApvK6fA70Xyzx2Yx4FMVzOTFPP/HE5xk4X4jRI8ZJnmneepN4wT92Yh0EGl9NQM88O9V4s88yVuXgZmR+0MEmj3J8Gn7uRP4LfOTTR7OWTpJk31LHt4MadBnB6JqYz8slY5A93x1ncEYZ4+D3NtVpKDxi9XO6OE7kjJilR0+ZTf4Y67Jy5UcdkjucZxnn1cI96u+Ns7igOxI8b0y0+TDzyuuN0DtRzeD1iZh7rkddtEzpJaRi6w/iAkqaeBqFYvpjVi0fLVexy26iL7pskw8/WASf/udvmdHL8yqPMdOIJut02o5PL0ZCbzp5P0O04qHmcu5ytYGpufrGn53acST1OXS5JwyjSQ3zIkulxt03o5OIkGz6f8gk2ecuMLpweU8sE7q/gZMsETvftAej0iFumb3o8jw2jKfrcMn3TP+Pm6xP8JbPM3fTUTWLAEx/g9J+5ZeqmJ+ymtR/sKXtPHIEwFuJikWYI5Xx6Lkekz6PMp78yR0iTI3Nft2MQu+XE3N89OASpOTH3d+MV0XhOzD3ebkfcoRN1j09ZEGHpRN3jwzXEkrpQ9/tQFXGzbuS9PUtHgLALc4/jJxAJ7cLc35AZRHy7EPc4Sgqh7S7MPQ6Mc5vHeRsJif9UcOPta+gr/iPDZYHu7/oc/3nisCXjMW+3TTiPiVumbuO2Xf3ljf+Xctpp95i5ZeI27nDFY96WedvY8zSPmX/ZmanHRG0xb38BH9ti3sbGQ3hM3RbzNjIKxmPmtpC3sbFPHlN3jHjzmLkt4G1snKO/1PGv407Rrf4yF/8mb41pFt/XXqfcPm/DW2oOTuODX4GB+lN8Xgrz+8TDz7T2iREf63Kll8XmEC2ybzz5Qn2kzyfIXBwLYtzKhj231LIHA70Y7V0fpuTR4R6V5k4DObylZqlr5/605h/L3bJOmSBT7WTTyELLdtMEmY9u319yUuoPfUsTnybZka18muRHN3SlmMWGf57aAkH8IW+L0p8g1ZGN/CVTr29+7nDXX7XZFv31PfsWN2tVoH9PiBuwxMVQWIHLeIZQbznxFBcB4aooGslj0ChTq6oszvavO9ovQIW95bjaR3aQWNwm9CwFhIGIM2FUOyVLbQWYGtQTzVTcdfQsBYylEMqJbq7PnbJc3D71HPZjCTCFn+kmlIa41Ok57MfijwOxYKNcEchS3B/lbp7L0AvGzPKBP7S+15GfpzMkosIAWQ/3qiE+CeSz1uP7gXNPtBTwZ1RkKvexh7AZ5VEIETDImSaS24ikgdwPNP0hxzVlz2GNhSjVR6RZFv9ZlQB3Ah8mSnrPk9ua0+M8HM2BScSrZnqDBTstBxXxWfSu9k/Dt/nhzfHBhXBKU+q8PrgjctdVj14TeXgdrrgb8lBVoDL3Ws5ez34HtB3Huk9XAAA=)

### Error
```
Render Error: <port#599(pin:1 .U1>.pin1) /> has multiple pcb matches, unclear how to place pcb_port: <smtpad#661(.1) />, <smtpad#721(.GND) />, <smtpad#722(.GND) />, <smtpad#723(.GND) />, <smtpad#724(.GND) />, <smtpad#725(.GND) />, <smtpad#726(.GND) />, <smtpad#727(.GND) />, <smtpad#728(.GND) />, <smtpad#729(.GND) />, <smtpad#730(.GND) />, <smtpad#731(.GND) />, <smtpad#732(.GND) />, <smtpad#733(.GND) />
```

seveibar, 2024-10-28:
@nischalj10 i also ran into this, we're working on it!!!

seveibar, 2024-10-28:
Related issue in the converter: https://github.com/tscircuit/easyeda-converter/issues/87

closed 2025-01-04 by github-actions[bot]