tscircuit/tscircuit.com #130: Profile page text update header

imrishabh18, 2024-10-28:
The Header is incorrect, it should be something like `My Profile` or `Profile`. And, the link redirected by the button `View Github` is redirecting to this url https://github.com/profile  and not the profile of the user.

[image]

SufyaanKhateeb, 2024-10-28:
I can work on this.

imrishabh18, 2024-10-28:
We are not assigning issues, until you have decent contribution in this repository. But feel free to work on this and open a PR

SufyaanKhateeb, 2024-10-28:
This issue seems to occur only if you navigate to the profile page using the footer link. This footer "My Profile" link is visible even if the user is not logged in, so what is the expected behaviour on clicking the link in that case or should I just hide the button in that case?

The footer link redirects you to the /profile if we just fix the redirect to go to the /{username} this problem will be fixed and also in this case the header text will be {username}'s profile, do you want it to be "My profile" or "Profile" if the user is in their profile page?

imrishabh18, 2024-10-29:
The profile link should not be visible in the footer when the user is not logged in.

cc: @seveibar

SufyaanKhateeb, 2024-10-31:
@imrishabh18 or @seveibar Can you please add the hacktoberfest-accepted label to this issue.

imrishabh18, 2024-10-31:
@SufyaanKhateeb Not needed, this repository has the `hacktoberfest` label. When you PR gets merged it will be counted towards the Hacktoberfest

SufyaanKhateeb, 2024-10-31:
It has been merged BTW. You can close this issue if it is OK.

closed 2024-11-01 by seveibar