tscircuit/tscircuit.com #171: Allow iframe embedding of snippets

seveibar, 2024-11-06:
Will require serving PreviewContent from a URL

the route could be `/preview?snippet_id=...` something like

No other page context except PreviewContent, automatically scaled to iframe size

seveibar, 2024-11-08:
/bounty $40

RohittCodes, 2024-11-08:
@seveibar if I understand correctly, this should be a separate page for previews only. like there is a `copy url` button for sharing editor, which routes to `/editor?snippet_id=...`

[image]

If so, can I provide a dropdown here, instead of copy url, with two separate buttons for share preview and share code/editor?

seveibar, 2024-11-08:
@RohittCodes from the "view snippets" page, users can get the iframe url. I'd prefer not to change that button on the editor page for now, I have some other plans for it

closed 2024-11-25 by imrishabh18