tscircuit/tscircuit.com #174: When running locally, automatically load all snippets from `@tscircuit/featured-snippets` module

seveibar, 2024-11-07:
The hardest part of this task is the typescript/javascript has to be compiled. This can be accomplished by importing `CodeRunner` from `@tscircuit/prompt-benchmarks`

/bounty $30

aybanda, 2024-11-08:
/attempt #174

RohittCodes, 2024-11-19:
/attempt #174

seveibar, 2024-12-01:
/bounty $50

ShiboSoftwareDev, 2024-12-16:
@seveibar can I instead fetch the snippets from the registry api? I see the featured-snippets repo was not updated for a month

seveibar, 2024-12-18:
@ShiboSoftwareDev hmm sort of creates an internet dependency when fetching locally- but it's kind of a good idea because we could also "load in problematic snippets". Maybe you could create a txt/json file like "autoload-snippets.json" that is filled with selected snippets, then people can also use it for reproducing issues!!!!!!

ShiboSoftwareDev, 2024-12-18:
@seveibar  I'll see what I can do

ShiboSoftwareDev, 2024-12-21:
I can now load selected snippets, should I also load all snippets from here: https://registry-api.tscircuit.com/snippets/list_trending ?

seveibar, 2024-12-21:
I wouldnt worry about loading trending snippets because youd have to load all their deps and it would be less predictable

ShiboSoftwareDev, 2024-12-22:
should I also recursively import all snippet's dependencies? 
[image]

seveibar, 2024-12-24:
Oof yea i guess so otherwise it wont work locally….

PR #423 (2024-12-26): import snippets from registry to development server
https://github.com/user-attachments/assets/0d5c5192-9087-434a-beb2-de3f2a798ecb

/claim #174
/closes #174
closed 2024-12-26 by ShiboSoftwareDev