tscircuit/tscircuit.com #175: Fix schematic zooming

seveibar, 2024-11-07:
When you zoom in on the schematic view, it doesn't zoom into the correct location. This most likely requires fixing `use-mouse-matrix-transform` (do not replace it- I don't want to get a bunch of mouse math into this project unnecessarily, but a PR fixing that project is acceptable for the bounty)

In this video I'm scrolling up and down, it SHOULD zoom into the mouse location but instead the screen swings somewhat wildly.

https://github.com/user-attachments/assets/e4627984-ddb9-4f0c-9cc7-996c8d55e45b

seveibar, 2024-11-08:
/bounty $20

RohittCodes, 2024-11-08:
Hey @seveibar, I would like to work on this. I see the issue is with the `CircuitToSvgWithMouseControl` component, but the actual `Schematic` component isn't stabilized and doesn't have the issue.

So, should I work with the `CircuitToSvgWithMouseControl` component?

RohittCodes, 2024-11-08:
/attempt #175

seveibar, 2024-11-08:
The Schematic component is too hard to maintain because it uses a more complex rendering engine, it doesnt have support for rendering circuit json properly, so the fix should be with the CircuitSvgWithMouseControl component

seveibar, 2024-11-08:
/bounty $30

PR #178 (2024-11-10): fix #175: zoom effect for schematic view
https://github.com/user-attachments/assets/c012a2bb-ac1c-4c19-ae47-6438f509b956

/claim #175

fixes: #175
commit 17256cbe (2024-11-10): fix #175: zoom effect for schematic view (#178)

* fix: schematic zoom

* chore: format

* refactor: bounds

* refactor: remove unused

* chore: formatting
closed 2024-11-10 by seveibar