tscircuit/tscircuit.com #200: Increasing the star duplicates on the dashboard

imrishabh18, 2024-11-13:
[image]

The `motor-driver-breakout` appears twice on the list

cc: @seveibar

ahmedhalac, 2024-11-13:
is this list of snippets hardcoded?

imrishabh18, 2024-11-13:
@ahmedhalac No, the trending snippets list is not hardcoded

Abse2001, 2024-11-21:
@imrishabh18 I can't repro in local
 [image]

imrishabh18, 2024-11-22:
@Abse2001 It is still getting duplicated as you can test in the https://tscircuit.com

Abse2001, 2024-11-22:
@imrishabh18 ya I was asking if you now if we can repro in local.

Abse2001, 2024-11-22:
when you are signed out there is no duplicate  https://tscircuit.com 
[image]
when signed in there is.
[image]

Abse2001, 2024-11-22:
@seveibar on local repo it does not get duplicated either if you are signed in or not idk how to repro this one.

imrishabh18, 2024-11-22:
Even I am not sure, probably it's a production api issue

seveibar, 2024-11-22:
@imrishabh18 yep it's a production API issue, @Abse2001 has figured out it's related to a join with user data because of the auth token

Internal issue w/ prod api filed here: https://github.com/tscircuit/registry-api/issues/36

Inside `getSnippetsQuery` function used here https://github.com/tscircuit/registry-api/blob/main/routes/snippets/list_trending.ts

closed 2024-11-28 by imrishabh18