tscircuit/tscircuit.com #214: Improve the error display when the 3d viewer fails to load

seveibar, 2024-11-15:
https://tscircuit.com/editor?&snippet_type=package#data:application/gzip;base64,H4sIABmtNmcAA5WTb2+bMBDG3/MpTn6VSIVg0q5pVapt7I82qd2ULtH2qgLHZUwBI9tUkxDffWdgxHQl6t7Zd7/n/NwdZHkppIYaKsXf3NxRSs/vl/dLaOBBihzIa61YtlD8kWdJLL0BeVy6kqfVPtZCEseuslFJNCVX+c6tVOKysWT9NfBP/SmRLE32iYKrTOHTJ+YSxWXMMryNK0hWZXrBhOSDNo+LKt6/32Va9aS36GIuN0HvlxIFcRz+u8WZKJSGzt2N2FV7DiHMSilKdYk+ijjnl6C0zIoUmjmE11A7MFIhP/Q3IxtK5gPQzqlNm9OMfLaTa552OWsnqA8sJKIdMXQ/IxElJ+iLdZG4YGiPUL/6gOEHIXSJTjWG/JV/RtDxoVbwb63T/6yFxSTXlSxghkeAq1SKqgRVJf0q7OmHtXVprlsBSvqhlSz5EdarBhZDpp2WYj+/h7VLg6ZH3BFjhtYjGO8Ivz2shY51JoqQ0JW/4ymxVBHtTwAFT0NScO19vH1Hhmgp0C7W9rafbpsh2j90MYoYS9Q7O8RGb1+0T9v8IRec20nLXfBid1823+jEy33XT82/Om7+uMG/DrWMGW//prA2WzL2aANaHMxa/dh0t21DvAzHJp9lp2CzLwN3trZvN3foa7p2N8Ch+hZ/uWNWDI4f4rP41aL9+M1l7jTOH086zrNlBQAA

[image]

/bounty $10

kom-senapati, 2024-11-15:
@seveibar I am not getting any 3d issue, how to trigger one

seveibar, 2024-11-15:
Hmm you're right its not there- i think it disappeared because my manual placements didnt save. We'll need to do a repro, ill post if i can repro again :/

PR #277 (2024-11-24): 3d-error-display
@seveibar cherry pick is cool

/closes #214
/claim #214
closed 2024-11-24 by ShiboSoftwareDev