tscircuit/tscircuit.com #221: Manual edits file doesn't save/persist, doesn't work on View Snippets page

seveibar, 2024-11-15:
CC @imrishabh18 can give more details

/bounty $40

hunxjunedo, 2024-11-15:
@imrishabh18 I'd like to work on this, can you please share instructions to reproduce this ?

imrishabh18, 2024-11-15:
- @hunxjunedo For this issue, as you can see in the video, that in the _view  snippets page_, in the files `manula-edits.json` is not there and because of that when you move components or edit traces it is not persisted. While the edit code page is having the persistance (currently there is some bug and even there the persistence is not working)

  [https://github.com/user-attachments/assets/9dfa7aa4-be21-466a-b9dc-a2bcfdc08993](https://github.com/user-attachments/assets/9dfa7aa4-be21-466a-b9dc-a2bcfdc08993)

hunxjunedo, 2024-11-15:
@imrishabh18 so I've successfully investigated looked into it, there are a few things I want discuss before writing the solution, can we have a call ?

imrishabh18, 2024-11-15:
@hunxjunedo Please join our discord, and paste your questions there. We can have a discussion over there

PR #248 (2024-11-20): Manual Edits to `snippets.manual_edits_json` and can be viewed in the files view
included manual-edits.json file and implemented update function for it.

[image]

fixes: #221 
/claim #221
closed 2024-11-20 by seveibar