tscircuit/tscircuit.com #222: The chip 3D view is missing the chip (regression)

imrishabh18, 2024-11-15:
https://tscircuit.com/imrishabh18/motor-driver-breakout

The 3D view is missing the chip, while it was present before.

[image]

seveibar, 2024-11-15:
@imrishabh18 is it a jlcpcb chip? This is likely because of an unavoidable CORS error when we use both snippets.tscircuit.com and tscircuit.com. To fix it, we have to add a url parameter that includes the domain to "bash the cache"

seveibar, 2024-11-15:
In effect, this means the fix has to be in core. Basically in `3d viewer` or `jscad-electronics` we need to append the domain as a query parameter whenever we request a model from the model cdn. It doesn't matter if the model cdn accepts the parameter or not- it's just to trick the browser into bashing the cache.

e.g. `modelcdn.tscircuit.com?.....&origin=snippet.tscircuit.com`

seveibar, 2024-11-15:
/bounty $15

imrishabh18, 2024-11-16:
@seveibar The error is not consistent, when I open the chip 3d view here - https://tscircuit.com/imrishabh18/TB6612FNG. I can see the chip rendering file.
But now the 3d view is failing for the motor-driver - https://tscircuit.com/imrishabh18/motor-driver-breakout

imrishabh18, 2024-11-16:
@Anshgrover23 Yeah, the errors in the 3d viewer are not consistent. The `jscad-fiber` version when updated in 3d-viewer was breaking the view. Cause the `createJSCADRenderer` in fiber is modifying the structure of jscadPlan.

I will create an issue in the `jscad-fiber` about this, if you want to take a look and tackle this. It would be great

imrishabh18, 2024-11-16:
https://github.com/tscircuit/jscad-fiber/issues/92

Take a look @Anshgrover23. I have added some of my findings while debugging

closed 2024-11-17 by imrishabh18