tscircuit/tscircuit.com #259: Add bundle size analysis comment as a comment on each pull request

seveibar, 2024-11-20:
I want to start tracking our bundle size. Create a github workflow that comments on each PR with the bundle size and gives a breakdown of the size. We have one in [tscircuit/core](https://github.com/tscircuit/core) but it's not purpose-built for vite applications.

/bounty $40

imrishabh18, 2024-11-21:
/bounty $40

imrishabh18, 2024-11-21:
@seveibar I think you need to look at the algora, as the `/bounty` is not working

algora-pbc, 2024-11-22:
## 💎 $40 bounty [• tscircuit](https://console.algora.io/org/tscircuit)
### Steps to solve:
1. **Start working**: Comment `/attempt #259` with your implementation plan
2. **Submit work**: Create a pull request including `/claim #259` in the PR body to claim the bounty
3. **Receive payment**: 100% of the bounty is received 2-5 days post-reward. [Make sure you are eligible for payouts](https://docs.algora.io/bounties/payments#supported-countries-regions)

Thank you for contributing to tscircuit/snippets!

**[Add a bounty](https://console.algora.io/org/tscircuit/bounties/community?fund=tscircuit%2Fsnippets%23259)** • **[Share on socials](https://twitter.com/intent/tweet?text=%247+bounty%21+%F0%9F%92%8E+https%3A%2F%2Fgithub.com%2Ftscircuit%2Fsnippets%2Fissues%2F259&related=algoraio)**

closed 2024-12-05 by imrishabh18