tscircuit/tscircuit.com #262: test footprinter default num_pins, and readonly params

ShiboSoftwareDev, 2024-11-20:
- since num_pins should be present in all footprints, remove the "Number of Pins" input field.
- make sure to figure out which params are readonly and not display an input field for them.

seveibar, 2024-11-23:
/bounty $15

ShiboSoftwareDev, 2024-11-26:
Remove number of pins because you should be able to change it from the input under 'Parameters'.
Now the Parameters are sometimes readonly and sometimes are not, this depends on the their zod type created in the footprinter repo, only readonly paramters should not be editable.

ShiboSoftwareDev, 2024-11-26:
@techmannih as I said for the input fields under 'Parameters' only some of them are readonly. For example: [image]
only the num_pins param is readonly(z.literal) and all the other params should be editable.

ShiboSoftwareDev, 2024-11-27:
> [@ShiboSoftwareDev](https://github.com/ShiboSoftwareDev) pease check, Now the num_pins behaves like the other parameters and it is not editable.
> 
>  20241127125057.mp4

that's not what I said. some paramters are readonly and some are not; you'd know from their type definition. look at the old 'Number of Pins' input field to know how to handle num_pins because it is a special case, regardless of whether it is readonly or not. If you find it difficult to understand the terminology I'm using send my comments to chatgpt, give it some context and ask it to explain in simpler terms.

siva222003, 2025-02-26:
/attempt #262

ShiboSoftwareDev, 2025-02-26:
@siva222003 this issue is deprecated, should've been closed

closed 2025-02-26 by ShiboSoftwareDev

siva222003, 2025-02-27:
Okay, but still the chip preview breaks whenever the readonly params are changed right?