tscircuit/tscircuit.com #284: `EidtorNav.tsx` mobile ui overflowing

Abse2001, 2024-11-25:
Try make it more responsive and maybe try to use another design for the `unsaved ` for mobile.
[image]

[image]

closed 2024-11-26 by anas-sarkez

imrishabh18, 2024-11-26:
https://github.com/user-attachments/assets/dc77b556-84f1-40f2-97c7-74b603f80bc0

Can you guys please confirm this has been fixed? Cause I can't see this responsive. Is it a issue on my mobile end 🤔

Cc: @anas-sarkez @Abse2001

anas-sarkez, 2024-11-26:
I've fixed the editor page.
I think its another issue 
[image]
[image]

imrishabh18, 2024-11-26:
Not sure know about the redesign of the mobile view,but I think we can create a issue which lists all these small responsive fixes here and there for mobile view. And I can add a bounty to it

Abse2001, 2024-11-26:
@ imrishabh18 You are right that a redesign is not a top priority.
I will open issues for any bugs I encounter.