tscircuit/tscircuit.com #326: "Unsaved Changes" is always visible, even after saving

seveibar, 2024-12-02:
[image]

/bounty $10

RohittCodes, 2024-12-02:
@seveibar @imrishabh18 not sure but maybe an issue in production. works fine locally
https://github.com/user-attachments/assets/fe38d4ef-5001-444c-84be-11304aea4baa

krushnarout, 2024-12-02:
/attempt #326

krushnarout, 2024-12-02:
> [@seveibar](https://github.com/seveibar) [@imrishabh18](https://github.com/imrishabh18) not sure but maybe an issue in production. works fine locally https://github.com/user-attachments/assets/fe38d4ef-5001-444c-84be-11304aea4baa

Correct! But the UX is bad after clicking on save button it still show unsaved changes for a while.

But I fixed it #328 

https://github.com/user-attachments/assets/a8005c2d-1752-4613-aa45-58308fad1fa5

seveibar, 2024-12-02:
It sounds like we need to validate that this works using the vercel preview url, if the issue still exists on production then the issue isnt resolved

RohittCodes, 2024-12-02:
/attempt #326

divanshu-go, 2024-12-02:
@seveibar I think we are not able to login on vercel preview url i have checked it see video for reference:
https://github.com/user-attachments/assets/3c6cb27f-ee37-4220-b223-35eddb660752

ArnavK-09, 2025-01-17:
any updates on this issue? is it still valid? that tests?

seveibar, 2025-01-17:
@ArnavK-09 thanks for following up on this but this issue had been fixed some time ago

closed 2025-01-17 by seveibar