tscircuit/tscircuit.com #346: Make playwright tests not flaky

seveibar, 2024-12-05:
I keep seeing broken playwright tests in PRs

I don't know why it's flaky, but we shouldn't be seeing that so often in PRs

/bounty $5

seveibar, 2024-12-05:
Yes if its idiomatic

Anshgrover23, 2024-12-05:
@seveibar the tests are flaky they somtime passed but sometimes they take more time due to UI Delays
Before:
[image]
After tracing each of them:
- Some failed due to timeouts because some tests are accesing more than 1 UI elements and due to this they are fail to do in required time.
[image]
[image]
[image]

Anshgrover23, 2024-12-05:
/attempt #346

PR #350 (2024-12-05): add retries for flaky test
close #346
closed 2024-12-05 by seveibar