tscircuit/tscircuit.com #348: Copid URL for the snippet is a bit huge

imrishabh18, 2024-12-05:
As the snippet gets bigger the snippet url will be huge too, as we passing the snippet data in base64 encoded format. Probably a shorter link with unique uuid (compressed data in db) or a url shortner would be better.

https://snippets.tscircuit.com/editor?&snippet_type=board#data:application/gzip;base64,H4sIAFBBUWcAA81Y627bNhT+76cg+Mvp6gslX4c4mGNlQ4cgC5zLMhRFIUt0rE2WNF3WFYbffYeXiBQd28qWoQmKkofmx/OdK2kH6yROc7RBRUbn1L+kPtqiZRqvEf4hz7ygk9G/aLBw03ZK/VZIfdzQIddFtjov8jyO9sES2NFa8C1V6F22mO0DZWu/VWSLlocbDS+OshzNVkFyHUSX7oKGGZqgTQOhJIjI9+gjdmz86b2QLS73Stlm8k9XTrnQYwv3s1m50GcL00EpD7g8LOUhl7ulPOIyKeUxl61SJl2+oCgRznGqOBFOctpXC5ylQzQMp+kQ7VjO0yFKMeFMHaKoEc7VGasFTtYZqQXOdn5xc3GrXMYJ6z6yOOP7D1dqRfhVecUSlJXbLMFYGWUJwspsS/DVYiXoKhstQZfg9wjfPqhlTtrpsuW5WrY576otNmeu22Jz5jLgW+RmiOeTzKpLuswhq9L4S5lYHwEH2oEEm8wfQC2fcD18yo7nE6AuRluOPTn25TiQ41COIzmO5QixkxOpjkX8feOTZDcPHlfP0mOhqVKZ30hy3/XvxQRSVozyaEhRMUqykJBilGShBMQoyU7nFz+KmX3/ZB9kKGPXoH/zMvbp0i3CHDVP0OSMV6TgfXdzPgOqssabmMn4pPxY9gu+QfWPJr75lWi7ZDPiu8S8iS8vHL4D/qU0L9IINWGK0OkidlMfrShz2AT37DYZrdcYfQn8fDXBZNTu95jsFnmcxkVO0wlm85YXxoWPz/ghcAwnnniL3yabFhm0+1vUOWs8fehBB5JzhNZuVCxdDzjQ9NpN86tivWCnTm/X9NG1rdE1LvdG7ppO8B1RK5CXIqCTTaWvbcsdmbd6mGy6lQVgNVILQLO6Q/LWtizjOE/SIAKX/LlMbEsj4C3mce7mQRxNNuPyEM1a4XJlA32c4IjmbZZxhkpim6xafdOSFqkuMaaDCuw5Pnyr+gCPuz59xBrb1yBb1baHaxX1zbjW8Ws1Jf4D1U3nHVRhRnMkrm/0rvO051TWsOBpbXmbnuD2nKAz1E7iDPPeKyy7P7+7wU98e9uSppZuUlWQQX2iZv41CTw3DL+y2vdREKFp6hdBFKMrF/5bwh6vSFMa5SgM1kEeRI8nOrlUnmRU4FyrQK00uoOurT4QWDfyAED+MCOyU3+7ARnu5E6vTkCKJAkDvZdAb2Ad9env99ADMLvLZpbV78krXfxtd0J30A1aH3gtN+zW0Btww14HaHa+lgOe6XhvwAPggyR0c+qv4pAaXsjj5HMIz5+KacC7ahqrbo0dO8cJAM+u0A1pD7Xd2cpN4FwvSL2QqlP5datjRiWmJsuU3eomTf2eewM8FzH0w3U9h7beAtOaTv1/uWrdH95A8H5zfch33uA/dH7RWzrcJuJT/Q01iwso3I1+BWovq53Xs/Ywcj24M5wgpR6vQGx4Y18F75T9jr/2Xb7WUPtEBOLnazMIWjMCO4A36X9OLPF23Q3wyz1ifts54JBqIgt/HHfHbv68zB9Vra/mDu0ZVXYVs6NbXXjuH7vVtBerbOkaZodueZv8e5bVJzt7EsJXmaM0tXvmhTS1grxNXY9mKI+Rl1LoJdU3mCpNtvc0Z5v5TznwCJSBRG34egt4WLkj8ChkPwlgUKKX9R4cMXADhjuCsQzMsAamV8XYwPcYBkzQIaSOGgNjjQ5heE8AkIEB+SjGdMFBcyTGNuwZ18AYbgPxKMRwAXwXPY6BqFcwdVwA4dAxPOEORwcMrjitjjUQQB1zMD0lxFBTI20ML5M6WsyKq5VpRqrVyQCz2pje4yAz1w4GVDjB1HMwOrLcDDUgQtM5gjHUcMxhiJGdpJarzZhqZbBHjZFqlsZsD8IoAeu4DjMuqmpOO/yXNSacNLaNfwAHgL8uLRgAAA==

cc: @seveibar

seveibar, 2024-12-05:
The base64/gzipped url is only suppose to be used for unsaved snippets, we could maybe have a dropdown for it in the future with a disabled option for copying by snippet uuid url if the snippet is/isnt saved. That said its very valuable to have base64 encoded snippet support- allows sharing for anon users and bug reports where the underlying snippet may be changed etc

imrishabh18, 2024-12-05:
Ohh, I didn't know it's for unsaved snippets only.

closed 2024-12-13 by seveibar