tscircuit/tscircuit.com #359: Developer Preview Login

seveibar, 2024-12-07:
We want to make it so you can login to preview urls

Logging in with github wont work on preview urls because they're not approved callback urls.

We should modify prod to allow vercel preview urls to login. This has some potential for security issues in the long term because anyone can deploy a vercel preview url and obtain a login token.

In the short term, this will be good for allowing us to test code that requires a logged in user.

My suggestion to make this more secure: add a toggle to the user settings page that allows a user to "turn on" developer mode with a warning about security. Only users with this mode on can login with vercel preview urls. We can also give developer mode some special feature access, such as the typescript debug files dialog (currently accessible to everyone always)

Cc @imrishabh18

seveibar, 2024-12-07:
/bounty $20

This issue maybe needs a bit more specification

hunxjunedo, 2024-12-09:
@seveibar is this one open ?