tscircuit/tscircuit.com #367: Sync to Github Repo Feature

seveibar, 2024-12-10:
We want to make it so we can sync snippets to a github repo the same way Vercel and Gitbook syncs with git repos.

- [ ] Ability to load in github repos
- [ ] Ability to write to github repos

This is a huge feature, I'm not going to bounty it because the setup is quite complex, but if you're interested feel free to propose a detailed implementation plan that could be bountied.

aybanda, 2024-12-10:
Hey @seveibar 

Implementation plan for GitHub sync (Vercel/GitBook style):

1. Repository Connection
- GitHub App installation flow (like Vercel)
- Repository selector UI
- Configure target branch and snippet directory
- Store connection settings

2. Two-way Sync
- Pull: Auto-load snippets when repo changes
- Push: Auto-commit snippet changes to repo
- Maintain sync status (like GitBook's sync indicator)

3. Core Features
- Watch for repository changes (webhooks)
- Handle concurrent edits
- Conflict resolution
- Sync status indicator

Initial focus:
1. GitHub App setup
2. Basic repo connection UI
3. Implement pull/push sync logic

Would you like to start with the GitHub App setup or break this into smaller issues?

seveibar, 2024-12-10:
@aybanda all of that is very obvious, I'm talking about an implementation plan specific to this repo. The plan would include endpoints to change, pages to create, and necessary schema changes.

aybanda, 2024-12-10:
Plan for github sync to snippets repo:

API Endpoints
POST /api/github/connect: Establish GitHub repo connection.
GET /api/github/snippets: Fetch snippets from GitHub.
POST /api/github/snippets: Push snippets to GitHub.
POST /api/github/webhook: Handle GitHub repo change events.

UI Pages
GitHub Connection Page: Setup GitHub App, select repo/branch.
Sync Status Dashboard: Display synced snippets and status.

Schema Changes
User Schema: Add githubConnection for GitHub details.
Snippet Schema: Add lastCommitSha for change tracking.

Sequence Flow
1. User connects GitHub repo.
2. App fetches/pushes snippets.
3. Webhooks trigger updates on repo changes.

aybanda, 2024-12-10:
sequenceDiagram:
    User->>+TSCircuit: Connect GitHub Repo
    TSCircuit->>+GitHub: Install GitHub App
    GitHub-->>-TSCircuit: Return Installation Token
    TSCircuit->>-User: Show Repo Selector

    Note over User,GitHub: Two-way Sync Flow
    
    GitHub->>TSCircuit: Webhook (Repo Changes)
    TSCircuit->>GitHub: Pull Changes
    TSCircuit->>User: Update Snippets

    User->>TSCircuit: Edit Snippet
    TSCircuit->>GitHub: Push Changes
    GitHub-->>TSCircuit: Confirm Commit
    TSCircuit-->>User: Update Sync Status

seveibar, 2024-12-10:
@aybanda still not at all close to a product spec.

Look at these:
- https://github.com/cake-build/rfcs/blob/master/templates/feature-template.md
- https://github.com/vuejs/rfcs/blob/master/active-rfcs/0028-router-active-link.md

RohittCodes, 2024-12-17:
@seveibar ig, this might be helpful! I have a few questions (listed them at the end) [Sync-Tscircuit-snippets-to-GitHub](https://steel-digestion-1a9.notion.site/Sync-Tscircuit-snippets-to-GitHub-15ff751ac5bf80a0b52cd0d119276cd0)

closed 2025-02-24 by github-actions[bot]