tscircuit/tscircuit.com #383: Wifi Camera Module

ShiboSoftwareDev, 2024-12-13:


seveibar, 2024-12-14:
that's a great one

seveibar, 2024-12-14:
/bounty $50

ShiboSoftwareDev, 2024-12-15:
I couldn't find this connector in jlcpcb, what should i do about it? I guess it has to be there for you to order it
[image]
[image]

seveibar, 2024-12-17:
Hmm can we find an alternate? Yea if JLC doesn't have it's a lot of effort for me to assemble- does JLC not have any camera connectors? I think they should be common

seveibar, 2024-12-17:
Can you give the part number? I may be able to help search for alternatives

ShiboSoftwareDev, 2024-12-17:
@seveibar I only have the name of the cam OV2640

ShiboSoftwareDev, 2024-12-18:
https://jlcsearch.tscircuit.com/components/list?subcategory_name=FFC/FPC%20Connectors these are the ones but I'm not sure which one we need, I only know it's a 24pin

seveibar, 2024-12-18:
Yea that's tricky- we need to get the standardized name of the connector or the pitch 🤔

seveibar, 2024-12-18:
Here are the dimensions from the Arducam Datasheet, we can use this to try to find a compatible connector 🏌 

[image]

[image]

seveibar, 2024-12-18:
@ShiboSoftwareDev I'm going to add you to the jlcsearch github repo, you might be able to introduce a page specifically for finding this type of connector (idk what to call it, "Flat Camera Connectors"?)

closed 2025-02-25 by github-actions[bot]