tscircuit/tscircuit.com #391: Programmable epaper Display

ShiboSoftwareDev, 2024-12-17:


ShiboSoftwareDev, 2024-12-22:
@Niharika0104 did you find a good board?

Niharika0104, 2024-12-22:
https://components101.com/module/1.54inch-e-paper-module
@ShiboSoftwareDev I found this can we go through this although this contains integration of e paper display with Arduino we can just skip the integration part and go ahead and design the schematic version of the module

ShiboSoftwareDev, 2024-12-22:
Looks good to me, @seveibar is this valid? Or do you have a specific thing in mind?

seveibar, 2024-12-22:
Yes anything from waveshare is good!

seveibar, 2024-12-22:
To be clear, you need to build a board that connects to that kind of display, no point in building the display board itself haha

ShiboSoftwareDev, 2024-12-22:
> To be clear, you need to build a board that connects to that kind of display, no point in building the display board itself haha

it's going to be connected to an arduino nano, which we already have a bounty for

ShiboSoftwareDev, 2024-12-22:
@Niharika0104 A lot of the components can be imported from jlcpcb, if not you can open issues for them

closed 2025-05-09 by github-actions[bot]