tscircuit/tscircuit.com #415: Make a quantity column to prevent superlong BOM

Abse2001, 2024-12-23:
This is what we have now.
[image]

I have this in mind. It will be clean, and we will keep track of the quantity of every part.
[image]

Notice that they share the same row when there is more than one name with the same part number.

cc @seveibar

seveibar, 2024-12-23:
@Abse2001 i like this, we should give it a name so circuit json to bom can optionally generate in this format

For example "withMergedQuantities" or something

Abse2001, 2024-12-23:
@seveibar should I open an issue in circuit-json-to-bom-csv repo for the withMergedQuantities ?

Abse2001, 2024-12-25:
@karthik-nair-20 ok go to this link to open the snippet for a live example 
https://tscircuit.com/editor?snippet_id=d5211bc2-835d-41cb-a10d-af0d8a04b8e0
Press RUN
[image]
open the menu and go to the bill of materials
[image]
you can see that we have two components with the same part number we want to combine the names like in the previous example and have a quantity column, this will be cleaner and prevent adding a row for every component with matching part numbers
[image]

Abse2001, 2024-12-25:
@karthik-nair-20 clone both snippet and https://github.com/[tscircuit/circuit-json-to-bom-csv](https://github.com/tscircuit/circuit-json-to-bom-csv) idk how is this generated excatly so you need to check snippet how the bill of materials work and I think you need to make the implementation in circuit-json-to-bom-csv repo you can then test your work locally with yalc 
you can then use
```
 supplierPartNumbers={{
        jlcpcb: ["<the part number>"],
      }}
```
to recreate the issue just add 2 componentes with different names with the same part number
[image]

seveibar, 2024-12-26:
/bounty $10

seveibar, 2024-12-26:
Note: The fix is inside circuit-json-to-bom as @Abse2001 mentioned, then you have to come back here and update the dep (or to tscircuit core)

seveibar, 2024-12-30:
This issue needs to be transferred to runframe

closed 2025-03-25 by github-actions[bot]