tscircuit/tscircuit.com #433: 5-3.3v regulator module snippet

ShiboSoftwareDev, 2024-12-27:


seveibar, 2024-12-30:
/bounty $10

yea we need a decent voltage regulator snippet, i suspect we may run into issues when actually _using_ the snippet, so the author of this should also demonstrate that it works inside of a larger circuit. 

I'm putting a bit lower of a bounty on this because i don't have a good quality criteria for it, and the regulator is typically only 4 chips or so, plus there are different power variants... eventually i would love to have a relatively standard module but i think you have to iterate on the design so it might not be well suited for a bounty.

The ideal circuit would be a `<group subcircuit />` which may have bugs, so whoever does this will likely get some good issues out of it.

seveibar, 2024-12-30:
The perfect version would actually swap components based on how much power you need etc. so you could say `powerRequirement="10W"` and it would select a good component. I'm not really sure what is best there but like i said something you iterate on and not super well suited to the bounty program

closed 2025-03-09 by github-actions[bot]