tscircuit/tscircuit.com #568: add support for 3D and assembly images in `fake-snippets-api/routes/api/snippets/images/[author]/[snippet_name]/[typeFormat].ts`

Abse2001, 2025-01-14:


seveibar, 2025-01-15:
/bounty $5

This is the prod backend implementation

```tsx
import { AnyCircuitElement } from "circuit-json"
import {
  convertCircuitJsonToPcbSvg,
  convertCircuitJsonToSchematicSvg,
} from "circuit-to-svg"
import { withWinterSpec } from "lib/with-winter-spec"
import { z } from "zod"

export default withWinterSpec({
  methods: ["GET"],
  auth: "none",
  routeParams: z.object({
    author: z.string(),
    snippet_name: z.string(),
    typeFormat: z.string(),
  }),
  rawResponse: true,
})(async (req, ctx) => {
  const { author, snippet_name, typeFormat } = req.routeParams

  // Check if the format is valid
  const format = typeFormat.split(".")[1]
  if (format !== "svg") {
    return ctx.error(400, {
      error_code: "invalid_format",
      message: "Format must be 'svg'",
    })
  }

  // Check if the type is valid
  const outputType = typeFormat.split(".")[0]
  if (outputType !== "schematic" && outputType !== "pcb") {
    return ctx.error(400, {
      error_code: "invalid_type",
      message: "Type must be 'schematic' or 'pcb'",
    })
  }

  // Get the snippet
  const name = `${author}/${snippet_name}`
  const snippet = await getSnippetsQuery(
    {
      name,
    },
    ctx,
  )

  // If snippet is not found, return 404
  if (!snippet) {
    return ctx.error(404, {
      error_code: "snippet_not_found",
      message: "Snippet not found",
    })
  }

  if (!snippet.circuit_json) {
    return ctx.error(404, {
      error_code: "circuit_json_not_found",
      message: "Circuit JSON not found for snippet",
    })
  }

  // Convert circuit json to svg
  let svg: string
  if (outputType === "schematic") {
    svg = convertCircuitJsonToSchematicSvg(JSON.parse(snippet.circuit_json!))
  } else if (outputType === "pcb") {
    svg = convertCircuitJsonToPcbSvg(JSON.parse(snippet.circuit_json!))
  } else {
    return ctx.error(500, {
      error_code: "unknown_error",
      message: "Unknown error",
    })
  }

  return new Response(svg, {
    headers: {
      "Content-Type": "image/svg+xml",
      "Cache-Control": "public, max-age=31536000",
    },
  })
})
```

ArnavK-09, 2025-01-17:
/attempt #568

closed 2025-01-27 by imrishabh18