tscircuit/tscircuit.com #582: improve error message for 3d viewer error for ux

ArnavK-09, 2025-01-17:
cc https://github.com/tscircuit/snippets/pull/562#issuecomment-2591639435

[image]

[x] willing to submit pr if issue valid

seveibar, 2025-01-17:
lol this is actually already the improved version of error message

We are slowly working on generating the 3d models server-side, so maybe we won't even have this error in the future

lmk if you have some ideas here

ArnavK-09, 2025-01-17:
> lol this is actually already the improved version of error message
> 
> We are slowly working on generating the 3d models server-side, so maybe we won't even have this error in the future
> 
> lmk if you have some ideas here
But I think this message should be something like " please switch to 3d view " as this error fixed after switching to 3d view, this can be checked by validating ref as undefined
This error message may make believe user that something internally is wrong and might not switch to 3d view

seveibar, 2025-01-17:
i guess i'd accept a PR for it but not compelling enough for a bounty imo

closed 2025-03-26 by github-actions[bot]