tscircuit/tscircuit.com #584: Error:  Invalid props for board unnamed manualEdits Expected object received string

ArnavK-09, 2025-01-17:
[Snippet code to reproduce](https://tscircuit.com/editor?#data:application/gzip;base64,H4sIAGS9imcAA72W34+aQBDH3/krNvukDyKg52lymKbau6SJl6ZpmvZxhVW2EdYsS68J8X/vyA9ZRGC9NH2TYT67853vMJGFRy4kSlES0w2XXKwF+00FOqGd4CHCH2TssTELBYsDsg3sufnt42xmO8+vL9go4JBEJDl88pmMC8ocQywhhxE9B81fMY+wYdA/WbpPd5Au0WCI3CVKDYQ8HsUSqde7V/UM8MbGQwNyBZWJiNAAfiL0tOVE+OiN+TJw8cQKQ4wCyvaBLJ9IAmfwRFLh4vPvkXfgiY/PNUN9Wc1uqgg4LbOD4Wi1nPElemRRQIlPRRFAKCIhdfHnLza+hCBpxZNIuun8VAW97Q83HS1qka9cEsl45KYLq3oRe8E5dVpFdpzLo2BwJIbDBX+bl7fplOZolaZfmVIYBKpMe66ktpcM74jHov2aCeplJD7QnWwq8sgREsGGK0UrpddFDok8eGGZdvKsiM1b/lgT9rMmvWi1Yz7UYpA0uSnFmlpKN2vqqz5pKJi0KLCtGwKU2toE2P+3fOWQfgP6y//n7S8FSEE8mi0lF5vwjaIlMmEa5xhJDpFVFojoHiui+5hJFuDxHcymDDjT91C2DvWYUxGV5vfVSgOYVcDL61oDeGjU9R7I0YBAbx2CU+6HQGAvNLmGoI33Q+BYLwS6r7qn074mpdM/OPna2hbCqc9c/zCUQDFuSmFdHpVUMXMK1WVSSTUHr8ulkmrMg73QoBreOl02lVTDJqfLppIqbFIouL3Nq3xXZXtHx6l8TWWrTSu92oTaG0vZhPpM2z0dO671ni6m9Z68xffd02VLbqZqS9f6zdNv2nIzv6pB8wutTLD0KroAxeRrX1B8X9r58PXdlW8rAlTgaZz96z8/DI2T8RccYMfywgwAAA==)

### Error
```
Render Error: Invalid props for board (unnamed): manualEdits (Expected object, received string)
```

seveibar, 2025-01-18:
This error irks me and is basically because the manual edits json file is empty by default.

Did you get this after forking? If so it's probably because when we fork we don't properly copy over the manualEdits file.

The hotfix for this is to put "{}" in the manualEdits file.

I would bounty this but I think it's going to get fixed in the runframe integration, so assigning to myself and sending you a tip, hopefully fixed tonight 😉 

/tip @ArnavK-09 $3

ArnavK-09, 2025-01-18:
> This error irks me and is basically because the manual edits json file is empty by default.
>
> Did you get this after forking? If so it's probably because when we fork we don't properly copy over the [...]

i didnt forked anything, i just clicked snippet on newest page

closed 2025-03-27 by github-actions[bot]