tscircuit/tscircuit.com #606: Cached duplicate subcircuit traces have incorrect hover

imrishabh18, 2025-01-24:
https://tscircuit.com/editor?&snippet_type=board#data:application/gzip;base64,H4sIABx2k2cAA42QvWrDMBSFdz3FRVMy2E7sKRCZQtstW5ausnRTC/RjZKktGL97FYWSUuyS7f6c7xzuVWZwPsAEJ5TPTkdjR5jh4p0B+hRGoaoRP1B13JcaZSFuEkpIVYFwdgzAY3DexYAeGEQr8aIsSrKwpNeu0E5wnQzwKycnPY86wGYLrIUNATh2jnsJn0qGntHDzhgKPar3PjBa5+7uyqZ7PbcJTvivSyw3yOh5T2EQ3RubimY3r9FQrfL1auQ/UPMT+khm8Fxgfjuj5XkP5en15QDloGyyCe46rPMwrdIwHbRM3kR/yGaBPFb5yy3Zkm8yI4YvAwIAAA==

On hovering over one trace, all the other duplicate subcircuit's similar trace light's up.

cc: @seveibar

Anshgrover23, 2025-01-24:
@imrishabh18 is this resolved??

[image]

imrishabh18, 2025-01-24:
Oops wrong url, I have update it. Thanks for pointing it out!

closed 2025-04-02 by github-actions[bot]