tscircuit/tscircuit.com #659: Closes the search box when clicking anywhere outside.

zalabhavy, 2025-02-09:
I want to add functionality like when you click outside the search box, it will automatically close and revert to the search button.

I Expect that you please provide me bounty of $10 for solve this issue.

 **Please assign it to me if this changes really needs because i have already solved it in my local system.**

 **Current**  :- 

https://github.com/user-attachments/assets/0de76866-e169-47e6-a05c-261182630e0b

**After my changes ** :- 

https://github.com/user-attachments/assets/3eb91b04-96f1-470f-a81e-40e75b128985

zalabhavy, 2025-02-09:
@imrishabh18 @seveibar please review this issue 😁

Thank You !

seveibar, 2025-02-10:
Not critical enough for a bounty assignment. Here's a good guide for literally getting hundreds of dollars in bounties https://docs.tscircuit.com/contributing/getting-started-as-a-contributor

(hint: start with footprinter, also join the discord and follow the bounties channel)

best of luck, here to help

closed 2025-02-10 by seveibar