tscircuit/tscircuit.com #674: Overlapping in header for small screen

zalabhavy, 2025-02-15:
Logo name tscircuit with Dashboard and Searchbox with contact overlapping for 1023x691 and smaller than that.

[image]

**I have solve this issue on my local machine**

https://github.com/user-attachments/assets/769fd012-3919-4bc5-8834-ce76edebede4

zalabhavy, 2025-02-15:
@imrishabh18 @seveibar Could you consider assigning a small bounty to this ?
It would motivate me to identify more issues like this that enhance the user experience on the official website. Even small improvements can have a significant impact.  

Thank you! 😊

imrishabh18, 2025-02-15:
@zalabhavy I am not able to reproduce this issue, the UI looks fine to me for that breakpoint. Also, we generally don't bounty UI fixes, unless it's very critical and breaking our UI lol. If you are looking for bountied issue you can check it out here - https://console.algora.io/org/tscircuit/bounties?status=open

closed 2025-02-15 by zalabhavy