tscircuit/tscircuit.com #733: Fix at least 2/3 of tscircuit.com SEO issues from ahrefs

seveibar, 2025-03-16:
/bounty $10

[image]

MustafaMulla29, 2025-03-17:
/attempt #733

Husainj, 2025-03-17:
/attempt #733

siva222003, 2025-03-21:
/attempt #733

## Implementation Plan : 

### Issue: Non-canonical page in sitemap  
The canonical URL is currently set statically, causing a mismatch between the URLs in the sitemap and their canonical versions. To fix this, we should dynamically set the canonical tag based on the current URL using the `react-helmet-async` package. Once implemented, we need to regenerate the sitemap to reflect the correct URLs.

### Issue: 3XX redirect in sitemap  
Some URLs in the sitemap are redirecting instead of pointing directly to their final destinations. We should update the sitemap with the final URLs to avoid unnecessary redirects. The affected URLs are:  

- [`https://tscircuit.com/playground`](https://tscircuit.com/playground)  
- [`https://tscircuit.com/community/join-redirect`](https://tscircuit.com/community/join-redirect)  
- [`https://tscircuit.com/legal/terms-of-service`](https://tscircuit.com/legal/terms-of-service)  
- [`https://tscircuit.com/legal/privacy-policy`](https://tscircuit.com/legal/privacy-policy)  

@imrishabh18 @seveibar Please check this and let me know if this is okay so I can start working.

siva222003, 2025-03-23:
@seveibar @imrishabh18, could you please confirm the implementation? If everything looks good, I will proceed with the work.

seveibar, 2025-03-31:
@siva222003 this ended up being a super hard issue, because it requires getting SSR to work or (what I ended up doing) patching the index.html via middleware

closed 2025-03-31 by seveibar