tscircuit/tscircuit.com #793: Error: Invalid props for platedhole unnamed shape Invalid literal value expected circle holeDiameter Requir

le51, 2025-04-02:
Hi,

I've started today to use tscircuit. It looks promising and fun. For now I've imported a couple of components using JLCPCB parts references (a really handy feature, great job) and added some decoupling capacitors.

But I'm stuck with this error with every kind of rotary encoder I try.

Exemple below is with: https://jlcpcb.com/partdetail/Bourns-PEC11R_4020FS0024/C143788

[Snippet code to reproduce](https://tscircuit.com/editor?#data:application/gzip;base64,H4sIAJsBqGcAAwMAAAAAAAAAAAA=)

### Error
```
Invalid props for platedhole (unnamed): shape (Invalid literal value, expected "circle"), holeDiameter (Required), outerDiameter (Required), innerWidth (Required), innerHeight (Required), rectPadWidth (Required), rectPadHeight (Required)
```

I've tried what is explained in [this closed issue](https://github.com/tscircuit/tscircuit.com/issues/584#issuecomment-2599576238) with no success.

seveibar, 2025-04-09:
Hey @le51 , sorry for the delay, I'm bountying this issue and we'll take a look soon!

/bounty $20

seveibar, 2025-04-10:
/bounty $0 cancelling bounty, we're working on it internally

andrii-balitskyi, 2025-04-13:
@le51 Hey, sorry for the delay. This is fixed now

closed 2025-04-13 by andrii-balitskyi

le51, 2025-04-14:
Hi, thank you very much.
I will give it a try.
Cheers.