tscircuit/tscircuit.com #840: `/editor` page doesn't support loading in package files, needs a file tree, needs to use package (not snippet) hooks

seveibar, 2025-04-13:
You currently can't edit anything other than manual edits and index.tsx

/bounty $50

[image]

seveibar, 2025-04-13:
/bounty $70

- [ ] File Tree
- [ ] Package Endpoint Transition
- [ ] `file_path` query param handling
- [ ] Selection of files
- [ ] Saving w/ new `package_release`

vishalpatil1899, 2025-04-13:
/attempt #820

seveibar, 2025-04-14:
The bounty on this is wrong, algora is adding instead of replacing 😭 the bounty is supposed to be $70!

MustafaMulla29, 2025-04-14:
> The bounty on this is wrong, algora is adding instead of replacing 😭 the bounty is supposed to be $70!

@seveibar now its multiple bounties😂

ArnavK-09, 2025-04-14:
Algora is a real dawg for that, respect

Anshgrover23, 2025-04-19:
@ArnavK-09 this one as well, always use FIXES #issuenumber in your prs so that this issue will automatically get closed.

closed 2025-04-25 by imrishabh18