tscircuit/tscircuit.com #866: Show error when the coordinates for the components are defined but manual-edits is imported

imrishabh18, 2025-04-17:
Similar to the error warning up on the code editor header when the `manual-edits.json` is not imported but has the edits details in it. We should also show error warning if the code has the `schX`, `schY`, `pcbX` or `pcbY` values but the `manual-edits.json` has the edit details. The warning should show an error message to remove the co-ordinates when using manual edits.

[image]

/bounty $5

MustafaMulla29, 2025-04-17:
/attempt #866

seveibar, 2025-04-17:
/bounty $25 requires a change to core to emit a warning

seveibar, 2025-04-17:
requires:

- [ ] Warning object defined in circuit-json
- [ ] Warning object emitted from tscircuit/core when there's a position conflict
- [ ] tscircuit.com displays helpful "in your face" error message
- [ ] PCB Viewer displays red box around component where two positions are defined

Do not claim until all 4 issues are resolved or your PR will be closed

ArnavK-09, 2025-04-26:
- [x] Warning object defined in circuit-json
- [x] Warning object emitted from tscircuit/core when there's a position conflict
- [ ] tscircuit.com displays helpful "in your face" error message
- [ ] PCB Viewer displays red box around component where two positions are defined