tscircuit/tscircuit.com #884: [Design Only] "Get Started" experience for tscircuit.com

seveibar, 2025-04-19:
Instead of just taking the user to the docs, I'd like to take them through a step by step process that explains tscircuit and has them change/execute code.

It should be sort of a "perfect introduction" to tscircuit. In the tutorial we could build the [usb c flashlight](https://tscircuit.com/seveibar/usb-c-flashlight)

This task requires a lot of design skill, I'm not sure exactly what it should look like. So the first issue here is just to get high-fidelity mockups (low-fidelity = excalidraw, high-fidelity = figma etc.) of the experience and copy (the text)

People will probably take elements of other designs and that's ok. I'll do partial rewards for people who inspired the correct design. You can use v0.dev etc. to help create the design.

Please post reference screenshots of great onboarding experiences!

/bounty $100

seveibar, 2025-04-21:
Example:

https://github.com/user-attachments/assets/63c44b3a-7636-4be6-ab41-00f7258d5f41

ArjunCodess, 2025-04-30:
https://21st.dev/originui/stepper/controlled-with-checkmarks
https://ui.aceternity.com/components/code-block
https://vercel.com/geist/code-block

amochuko, 2025-05-14:
/claim #884 https://www.figma.com/proto/JTIbqvTEzZ08xbfL7wWCdp/WorkHouse?node-id=667-282&t=RC82ISnRWBAHkfs5-1&scaling=scale-down&content-scaling=fixed&page-id=666%3A279

amochuko, 2025-05-14:
Hello @seveibar, I would like to submit a prototype at https://www.figma.com/proto/JTIbqvTEzZ08xbfL7wWCdp/WorkHouse?node-id=667-282&t=RC82ISnRWBAHkfs5-1&scaling=scale-down&content-scaling=fixed&page-id=666%3A279 for your review.

I would appreciate your feedback.
Thanks.

ArnavK-09, 2025-05-14:
> Hello @seveibar, I would like to submit a prototype at https://www.figma.com/proto/JTIbqvTEzZ08xbfL7wWCdp/WorkHouse?node-id=667-282&t=RC82ISnRWBAHkfs5-1&scaling=scale-down&content-scaling=fixed&page-id=666%3A279 for your review.
> 
> I would appreciate your feedback.
> Thanks.

[image]