tscircuit/tscircuit.com #920: Having a settings tab on the view page

imrishabh18, 2025-04-22:
We should be having a tab or some button similar to Github but which blends with this design for `settings`. Configurations of the package should be there, like changing the visibility private/public or delete (the most important ux, I feel is missing). And it should be shown to authorized users.

[image]

Share some designs which here, and then we can create an issue for it.

_Originally posted by @imrishabh18 in https://github.com/tscircuit/tscircuit/discussions/594_

imrishabh18, 2025-04-22:
/bounty $10

mohan-bee, 2025-04-22:
/attempt #920

imrishabh18, 2025-04-22:
Share some designs before making a PR, so that the loop of reviewing PR is faster!

mohan-bee, 2025-04-22:
My Intuition is to add a Settings icon where you have mentioned which will trigger a popup and the popup contains the change visibility and delete package.

@imrishabh18  How does it sounds !

closed 2025-04-25 by seveibar