tscircuit/tscircuit.com #990: Error:  Invalid props for capacitor C1 capacitance Required

harrington40, 2025-04-29:
[Snippet code to reproduce](https://tscircuit.com/editor?#data:application/gzip;base64,H4sIAMdiEWgAA41TTW+CQBS8+yte9qSpH4BajXFNWjRtUtODrY1N0wPiWjYRlsA2tSH89z6WFaHG2gt583Z2mJkA24cikuCKIJZwvxqY5q1wog1QqDeATqBeAxiv1eqLb6RHSc/wfQIe4x+epKSrkPiUOx4wmrwhHSDZj8Bowjc+02ax6RUr+L3rlnnFCjfv6UTtx67HQwgcn1GyNAlshZBhxAN0EAvuDgmE7npFians4PyKc1/NPJg7a7aLaZIoKQBzBGR2RfI3AlgZbBHtCqCL+Aaxhr0MHtn96ul19XSA8Ml+KPAQ8fSZKJSm0MnT6EhO6Lhcikjnsqu5jKHRP8SyyrHUrJVORax/ieRzxY6MHJfBNhI+Je2lCW2sCKRAYCPAGtHe4a2n3FaViy7OcrEuzbUu6mLTFe5RN7+wYxsdfD6bolCeNP8oq3XpCxGLeXysa3G+87LIhboCJtsvtq2tZk7+DLYojjN+dvnucaq544762Sa1xg+pIOi1mgMAAA==)

### Error
```
Render Error: Invalid props for capacitor "C1": capacitance (Required)
```

seveibar, 2025-05-02:
To be fair, this shouldn't cause the entire circuit to fail to render, it should just show up as a regular error

imrishabh18, 2025-05-04:
@seveibar Correct! We should add these errors to the circuit-json and not make this fail the entire circuit

closed 2025-05-12 by imrishabh18