tscircuit/tscircuit.com #1024: Can't download fabrication files from editor

seveibar, 2025-05-05:
[image]

CC @ArnavK-09 

/bounty $10

Anshgrover23, 2025-05-05:
@seveibar  i am able to download these files?

[image]

seveibar, 2025-05-05:
From the editor page @Anshgrover23

ArnavK-09, 2025-05-05:
On it

ArnavK-09, 2025-05-06:
ah different circuit jsons are being used

ArnavK-09, 2025-05-06:
```ts
var getApertureConfigFromPcbSolderPaste = (elm) => {
  if (elm.shape === "rect") {
    return {
      standard_template_code: "R",
      x_size: elm.width,
      y_size: elm.height
    };
  }
  if (elm.shape === "circle") {
    return {
      standard_template_code: "C",
      diameter: elm.radius * 2
    };
  }
  throw new Error(`Unsupported shape ${elm.shape}`);
};
```

ArnavK-09, 2025-05-06:
is this fabircation lib err?

ArnavK-09, 2025-05-06:
related https://github.com/tscircuit/tscircuit.com/issues/942

ArnavK-09, 2025-05-06:
tldr
`circuit-json-to-gerber` doesnt support pcb_solder_paste of pill type which is being generated in new core versions

[image]

[image]

ArnavK-09, 2025-05-06:
ref commit https://github.com/tscircuit/core/commit/acd3e866591c3c27c071905797889bc3b58268f4

ArnavK-09, 2025-05-06:
Or should I try circuit-to-gerber to support pill?

seveibar, 2025-05-06:
@ArnavK-09 let's get circuit json to gerber to support pill, going to upgrade the bounty to reflect that

/bounty $20