tscircuit/tscircuit.com #1058: Seeing the error warning by default even for default pacakges

imrishabh18, 2025-05-08:
[image]

This shouldn't be showing up for packages which are not using the manual-edits

ArnavK-09, 2025-05-08:
Yup because in your recent pr, you added `{}` content to default manual edits 
Either we need to revert default to ` `
Or check for empty json
Which is the way?

imrishabh18, 2025-05-08:
I think manual-edits.json file should have `{}` cause we filter the file in eval. Having it empty doesn't feel right

cc @seveibar

seveibar, 2025-05-08:
Great discussion- I think @ArnavK-09 is right- it should not be an empty object. manual-edits.json should have no special handling with eval at all, it's not going to be the only json file with data (we will eventually have stuff like manual trace files, and people will do things like customize the path to the manual-edits.json file)

So we should remove the code that makes it an empty object, and adjust the error/warning.

seveibar, 2025-05-08:
new user experience is really important btw, so this is a good find @imrishabh18

imrishabh18, 2025-05-08:
Okay, we are moving back to empty file then we should make sure that the circuits which already have manual-edits.json don't throw error of file not found.

ArnavK-09, 2025-05-08:
> Okay, we are moving back to empty file then we should make sure that the circuits which already have manual-edits.json don't throw error of file not found.

Is something about that updated in eval or core? Previously it wasn't throwing such errors with blank or filled manual edits

seveibar, 2025-05-08:
Just noting that there's actually two unusual behaviors. `manual-edits.json` shouldn't be created/available unless the user manually edits something

[image]

ArnavK-09, 2025-05-08:
> Just noting that there's actually two unusual behaviors. `manual-edits.json` shouldn't be created/available unless the user manually edits something
> 
> [image]

Ignore error would be fixed in my pr,
Manual edits option is given intentionally if user wants to add custom manual edits having saved without using pcb viewer or schematic, do I need to remove it?

seveibar, 2025-05-08:
@ArnavK-09 yea i don't think the user will manually create that file (ironically given the name)

PR #1032 (2025-05-08): refactor(CodeEditorHeader): logic to check if manual imports not used also add support for old spec file 
now no default manual edits, only when updated from runframe, 
fixed editor loading state logic
fix #1058

https://github.com/user-attachments/assets/94aebbc3-a6a7-4987-81d2-1ba64913d988

…,

The `checkIfManualEditsImported` function was updated to accept a `file` parameter to check for manual edits in a specific file instead of hardcoding "index.tsx". This ensures the function works correctly for different entry points and file types.

now it checks both

```ts
import manualEdits from "./ [... cut]
closed 2025-05-08 by imrishabh18