tscircuit/tscircuit.com #1494: Error Execution Error

Vizion-HUN, 2025-07-20:
[Package code to reproduce](https://tscircuit.com/editor?#data:application/gzip;base64,H4sIAJsBqGcAA8vMLcgvKlFwyk8sSlFIK8rPVVDS008C8fRKiiuUuJIzi5JLM0v0ElNSNGwgqvTtNAFBGkttNgAAAA==)

### Error
```
Execution Error: Eval compiled js error for "entrypoint.tsx": Unsupported component type (not registered in @tscircuit/core catalogue): "header" See CREATING_NEW_COMPONENTS.md
@tscircuit/eval@latest
Error: Eval compiled js error for "entrypoint.tsx": Unsupported component type (not registered in @tscircuit/core catalogue): "header" See CREATING_NEW_COMPONENTS.md
    at importLocalFile (blob:https://runframe.tscircuit.com/b38301a5-4722-4157-815a-e8884ba55052:269631:13)
    at async Object.executeWithFsMap (blob:https://runframe.tscircuit.com/b38301a5-4722-4157-815a-e8884ba55052:269973:5)
```

Weird, bleeding hard at the first component... it is useless, sorry

imrishabh18, 2025-07-20:
It's `pinheader` not `header`, take a look at the docs for the component names.

Vizion-HUN, 2025-07-23:
it is not about pinheader vs header.
We started with exact parts 3p terminal: it founded the part, but instead of placing, just punched holes in the PCB. Then I asked for standard pcb to wire connector, any type, and asked several others to solve the errors.
Does not work. So if it has no synonims for parts, and can not lookup exact part numbers or websites...it is pretty long journey to create a pcb.

Vizion-HUN, 2025-07-23:
did not import, just told pthe part number and the details, as it asked.

seveibar, 2025-07-24:
Sorry @Vizion-HUN , I'm not sure i see an MPN anywhere in the messages above. An MPN looks like this: RP2040, PZ254V-11-02P, or ZX-PZ2.54-2-6PZZ etc.