tscircuit/tscircuit.com #1691: Error Execution Error/Couldn't link my cloned core repo

BenraouaneSoufiane, 2025-09-21:
[Package code to reproduce](()=>{if(e)try{return VUt(e instanceof Map?Object.fromEntries(e.entries()):e)}catch{}return Y0n(n??"")})

### Error
```
Execution Error: this.getSubcircuit(...).getNextAvailableName is not a function
@tscircuit/eval@0.0.333
TypeError: this.getSubcircuit(...).getNextAvailableName is not a function
    at mx.doInitialAssignNameToUnnamedComponents (blob:http://localhost:3020/e6d70d59-5ea6-483b-98b1-d34a2a5b05c4:226:13603)
    at mx.runRenderPhase (blob:http://localhost:3020/e6d70d59-5ea6-483b-98b1-d34a2a5b05c4:226:1853)
    at cY.runRenderPhaseForChildren (blob:http://localhost:3020/e6d70d59-5ea6-483b-98b1-d34a2a5b05c4:226:2002)
    at lbt.runRenderPhaseForChildren (blob:http://localhost:3020/e6d70d59-5ea6-483b-98b1-d34a2a5b05c4:226:1971)
    at Jj.runRenderPhaseForChildren (blob:http://localhost:3020/e6d70d59-5ea6-483b-98b1-d34a2a5b05c4:226:1971)
    at Jj.runRenderCycle (blob:http://localhost:3020/e6d70d59-5ea6-483b-98b1-d34a2a5b05c4:226:1111)
    at Hp.render (blob:http://localhost:3020/e6d70d59-5ea6-483b-98b1-d34a2a5b05c4:227:58973)
    at Hp.getCircuitJson (blob:http://localhost:3020/e6d70d59-5ea6-483b-98b1-d34a2a5b05c4:227:59590)
    at Object.getCircuitJson (blob:http://localhost:3020/e6d70d59-5ea6-483b-98b1-d34a2a5b05c4:388:3283)
    at i (blob:http://localhost:3020/e6d70d59-5ea6-483b-98b1-d34a2a5b05c4:130:1963)
```

seveibar, 2025-09-21:
@BenraouaneSoufiane what was the package code to reproduce? It looks a bit odd

BenraouaneSoufiane, 2025-09-21:
> [@BenraouaneSoufiane](https://github.com/BenraouaneSoufiane) what was the package code to reproduce? It looks a bit odd

I've three days on this, just want to use my local forked coloned repositories which includes the implementation of boxed subcircuit to create the svg image you were requested in my previous PR, to summarize I wanted to link repositories then using the tsci to reproduce the SVG

BenraouaneSoufiane, 2025-09-21:
I want from tsci to use doInitialAssignNameToUnnamedComponents which comes from my cloned core repo not the latest version of core comes from registry or whatever, if possible tell me how to use my cloned core repo in the tsci

closed 2025-11-29 by github-actions[bot]